Maritime-Robotics-Student-Society/sailing-robot #156: Simulator Question

soulslicer, 2016-10-21:
I would like to enquire if your simulator is using Gazebo? and it's doing fluid simulation etc?

takluyver, 2016-10-21:
No and no, as far as I know. It's a simpler kind of simulation - it has a lookup table of speeds when the wind is blowing at different angles relative to the boat. It's there to test that our code is doing something sensible, not to estimate the boat's performance.

You can see the code calculating the simulated velocity [here](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/blob/master/src/sailing_robot/scripts/simulation_velocity). There are other simulation components in the same directory.

tsaoyu, 2016-10-21:
Please have a look on Wiki page: 
 https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Simulator
@Nanoseb designed and implemented the simulator. 
____________________________

(* This issue was closed because no question or suggestion has been raised in this conversation. But you are welcome to continue this discussion here if you got any other questions. *)

closed 2016-10-21 by tsaoyu