Maritime-Robotics-Student-Society/sailing-robot #158: Update high level node overview

tsaoyu, 2016-11-24:
Update our high level node overview graph with launch files and `rosgraph` tool

ukcojamak9-martin, 2016-11-26:
I am trying to get a picture of what is going on using rosbag, but I am having problems.

When I play a bag file with " rosbag --play 'ros_file.bag' ", initial indications are that it is working: " rostopic list" shows the topics and I can see the messages via " rostopic echo 'topic' " even the rospy command line interface I have written to help me understand what the boat is doing can see the messages.

However " rosgraph ", " rqt_graph " and " rviz " all appear to not be seeing what is going on.

When I " roslaunch simulator.launch " everything seems to work normally, which is all I have needed so far.  But to update the node overview graph, it would be better if I could get "rosgraph" and "rqt_graph" working with a bag file so that I can see the actual nodes rather than the simulation nodes.

Anyone got any idea what I am doing wrong with rosbag please?

In case it helps: ROS is being run inside a Docker container.  I am running commands through a direct bash console for the CLI commands and X-forwarding through SSH for the graphical interfaces.

tsaoyu, 2016-11-28:
My assumption is `rosbag` we had didn't record all topics during test. 
Take a look what topics are included with `rosbag info <your bagfile>` command.

ukcojamak9-martin, 2016-11-28:
Thank you, I have just tried that, and it shows that the topics are there with a load of messages (I am using the extravaganza.bag file from the git-challenge repository).

This if confirmed by me being able to see the messages as they are being replayed by rosbag in some programs like " rostopic echo <topic_name> "

smaria, 2016-12-07:
It may have something to do with the order in which you launch things. I usually start a roscore, then launch a bag file (whilst playing, try the --loop option so you don't have to keep relaunching), and finally the tools that I want to use.

ukcojamak9-martin, 2016-12-09:
Sorry for not updating this, We worked out what was wrong on the software meeting last week.  Ii was only showing one node as there is only one node issuing all the messages when running a bag file.

I will need to get a copy of the picture off of the Raspberry Pi when it is actually running ROS

smaria, 2017-02-02:
I can't quite follow how this is needed for a node overview?
What information are you looking for?