Maritime-Robotics-Student-Society/sailing-robot #159: Look into compass calculations without calibration

takluyver, 2016-12-01:
The algorithm described in this paper can apparently give heading and orientation without needing calibration:

www.mdpi.com/1424-8220/15/8/19302/pdf

tsaoyu, 2017-05-01:
In this new IMU [driver](https://github.com/tsaoyu/RTIMULib), it use Kalman filter to get fused magnetometer, gyro and accelerometer reading. I have implemented this on our new RPi3, it gives a very stable yaw angle with careful ellipsoid magnetometer calibration. Based on this, I would like to do some further tests: 
 * How battery pack position influence *fused* orientation reading?
 * How our boat influence *fused* reading? Does it possible to use bare RPi + IMU calibration profile for the boat
 * How stable this calibration is? Can we detect abnormal magnetic filed change?

tsaoyu, 2017-05-01:
[RTIMULib2](https://github.com/RTIMULib/RTIMULib2) said to have runtime magnetometer calibration functionality 

> RTIMULib2 is the second version of the original RTIMULib library. The main change is the addition of the runtime magnetometer calibration functionality.

If we found all three issues we mentioned before all have negative answers, runtime calibration is still an option for us.

takluyver, 2017-05-09:
@NanaOKA just pointed me at [imu_complementary_filter](http://wiki.ros.org/imu_complementary_filter), which is written by the author of the paper I linked to above (Roberto Valenti). I don't know how it compares with RTIMULib2.