Maritime-Robotics-Student-Society/sailing-robot #160: Improvement of Dashboard

tsaoyu, 2016-12-01:


takluyver, 2016-12-01:
[Rosbridge](http://wiki.ros.org/rosbridge_suite) provides a websocket interface to ROS; this could be more flexible than our websocket server which publishes a fixed list of topics.

tsaoyu, 2016-12-01:
https://github.com/pantor/ros-control-center

takluyver, 2016-12-01:
@SuperWool I remember you were reading about JS development - you might be interested in looking at these things.

smaria, 2016-12-07:
rosbridge: looks good on ros.org, but the website linked is now no longer owned by them. the mailing list looks dead-ish, but might just be complete in what it does?

is there a ros.org page for ros-control-center? 
I am not sure how much we can gain from it, the views they are offering are not really what we need; 
Is there anything more they make available in addition to rosbridge that we can build our own visualization on?

takluyver, 2016-12-07:
Rosbridge: http://robotwebtools.org/ still looks like it's run by the same group, at a glance, and https://github.com/RobotWebTools/rosbridge_suite appears to be updated.

ZhongYuenLee, 2017-08-31:
Done it sorta

takluyver, 2017-09-10:
Thanks - I think we can close this issue and open more specific ones.

closed 2017-09-10 by takluyver