Maritime-Robotics-Student-Society/sailing-robot #161: Better way of debugging

tsaoyu, 2016-12-01:
Find a better way/ toolset to debug our code.

tsaoyu, 2017-05-01:
I spotted this VS code based ROS development IDE http://www.roboware.me/

smaria, 2017-07-22:
What are we currently missing for debugging? What current problem are you trying to solve with this new tool? This is not a very specific issue.

takluyver, 2017-07-25:
Yeah, sorry, this isn't a very useful issue. I think we created this when we were discussing my frustration with trying to debug behaviour problems with the boat - it's hard to debug during a test because you're short of time, and trying to do everything on a laptop balanced on your knees while someone shades it from the sun. But when you get back, it's tricky to remember what was going on and find the relevant information to make sense of it.

There were a couple of concrete things that came out of this discussion:

1. The index of data files from tests, to make it easier to find the file you're looking for.
2. The system for recording timed notes, so we have a quick way to record when something is happening which we might want to look at later.

Maybe I'll have a go at writing a blog post about these tonight, if you like. In the meantime, I'll close this issue - we should open more specific issues if there are particular things we want to improve.

@tsaoyu thanks for the link to roboware, it looks interesting.

closed 2017-07-25 by takluyver