Maritime-Robotics-Student-Society/sailing-robot #162: Correct inconsistence of convention system

tsaoyu, 2016-12-01:
There are some inconsistence between convention system and our code. 
We need update them on WiKi page.

tsaoyu, 2016-12-05:
In more detail we need document the following:

- Zero position and its reference
- Direction convention of signs
- Unit type

We can use hardware to check the consistency of conventions with real hardware. 
[Servo test](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/blob/2c719e66094c148de513542f2b8a181f4a824b80/src/sailing_robot/scripts/actuator_driver_rudder) script is reusable to test sail angle/ rudder angle.

smaria, 2016-12-07:
I quite liked the proposal at the southampton python user group, of having classes that return angles etc. in different reference systems.

A lot of conventions are already documented on the [naming conventions](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Naming-conventions#wind_direction_apparent) page, but may need updating...

Which conventions are we missing from that page?