Maritime-Robotics-Student-Society/sailing-robot #166: Docker file

tsaoyu, 2016-12-01:
Docker is probably a good idea to create portable ROS image for us. 

@ukcojamak9-martin create a general docker file that useful for our project.
(Can you share docker and put some document in https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Virtual-Machine-for-Windows-Mac-users ?)

And we want to know for Raspberry Pi, are there any restriction on hardware access? 

Reference: http://blog.hypriot.com/

ukcojamak9-martin, 2016-12-03:
Guidance on stalling Docker and a sailing-robot image are now in the wiki.  dockerfile files have been pushed to the repository.

takluyver, 2016-12-03:
Thanks @ukcojamak9-martin !

Leaving this open for now to discuss the tradeoffs of using docker on the raspi as well.

tsaoyu, 2016-12-03:
Thanks Martin! 
I have tried docker on macOS.  The isolated docker developing environment seems promising to me.
And here are some bugs I found when I make it works on macOS.

1. Shared `catkin workspace` only works on Linux
```
docker run --name sailing-robot -h sailing-robot -v /Users/caoyu/PycharmProjects/sailing-robot:/home/pi/sailing-robot --rm -it jamak9/sailing-robot:indigo

indigo: Pulling from jamak9/sailing-robot
Digest: sha256:f3fe5cc108f20e9ec4f1224a09330fa9c0d2a53db60b23f90e2fa1c0bd6830dc
Status: Downloaded newer image for jamak9/sailing-robot:indigo
```
And the error message: 
```
bash: /home/pi/sailing-robot/devel/_setup_util.py: /usr/local/bin/python: bad interpreter: No such file or directory
Failed to run '"/home/pi/sailing-robot/devel/_setup_util.py" ': return code 126
```
This is because ROS was installed on mac with `homebrew` version of python. So the python interpreter is located at `/usr/local/bin/python`, cause error when linking host and container. 

I think it would be great if we can point out that "Shared folder allows easy access to source file but the compatibility is based on host environment. " For me I have completely give up running ROS on macOS due to regular crash of `rviz` and `rqt`. And I just wondering, does it possible to `git pull` in the container and keep it completed isolated with host? 

2. Terminal access to container works well

`docker exec -it sailing-robot bash` 

3. No lucky with `ssh` access and X-forwarding

After setup `sshd` in container, I have tried `ssh pi@172.17.0.2 ` but time out.

And also `ping 172.17.0.2` with result 
```
PING 172.17.0.2 (172.17.0.2): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
Request timeout for icmp_seq 4
Request timeout for icmp_seq 5
Request timeout for icmp_seq 6
```

I am investigating this issue to make sure we can run GUI program like `rqt` and `rviz` with X-forwarding on host computer. It would be helpful if you can share the output on yours computer of the two commands I have tried.

ukcojamak9-martin, 2016-12-03:
1) If you remove `-v /Users/caoyu/PycharmProjects/sailing-robot:/home/pi/sailing-robot` from the `docker run...` command, then `git clone`, `git pull` and all other git commands should work as normal.  However, everything that you do will be forgotten when you shutdown and restart the docker container.

You can also change it to this instead:: 
`-v sailing-robot-git:/home/pi/sailing-robot`

It will appear very similar to removing it totally, but it should remember the files within /home/pi/sailing-robot between restarts. (It does this by creating a local container to contain the files within that directory.)

I have just tested this on linux and once started I had to `chown` the /home/pi/sailing-robot to pi as it defaulted to root.  It did remember the owner on restart though.

2) good.  :-)

3) Assuming that the ssh server started properly, then my next step in diagnosing it would be to check the ip address, as the one in the script is the default one on my computer, it may be different on a mac though.
Not sure how to do that on a mac, but on linux it is with `ifconfig` within the container which comes up with two interfaces eth0 and lo  it is labelled 'inet addr:' under the eth0 interface.

(Sorry if I am telling you stuff you already know, also if there is not enough detail please let me know.)

ukcojamak9-martin, 2016-12-03:
Just been thinking a bit more.  Before you edit the `docker run...` line:

 within the container try running:
`sudo ln -s /usr/bin/python /usr/local/bin/python` 

to see if that helps.

If it does, I will add it to the docker image.

ukcojamak9-martin, 2016-12-03:
I have now uploaded a new version of the docker image to, hopefully, fix the python location problem on mac's.

Please `docker pull jamak9/sailing-robot` and let me know if it has fixed it please.

tsaoyu, 2016-12-04:
Thanks, Martin. I can confirm the symbol link for python interpreter works well!

ukcojamak9-martin, 2016-12-04:
How is SSH going, did you manage to check the ip address?

tsaoyu, 2016-12-04:
I have managed to `ssh` into `sailing-robot` container with bit of effort
[Here](https://docs.docker.com/docker-for-mac/networking/#/i-cannot-ping-my-containers) I found it is not possible to `ping` into containers on macOS. As a work around I publish the ssh port 22 to 2222 with command
```
docker run -p 2222:22 -it jamak9/sailing-robot
```
and then inside the container start ssh service.

And the `ssh` access is done by
```
ssh -p 2222 pi@localhost
```

I am still working on X window forwarding issue. But I am not really optimistic about the result,  [rviz](http://answers.ros.org/question/27695/x11-forwarding-and-ros-visualization-tools/) doesn't play well with X window. To have a fully functional desktop development environment, I would suggest using native `homebrew` approach.

ukcojamak9-martin, 2016-12-04:
Can you update the wiki with Mac instructions please, as you understand it much more than me. `rviz` works well for me on X windows, so maybe it is a mac issue?

I know nothing about homebrew, so can not advise on this.

takluyver, 2017-01-26:
I've added the three scripts from the wiki into git, and made the one that starts the docker container a bit smarter to find the repository directory by itself.

closed 2017-10-10 by Nanoseb