Maritime-Robotics-Student-Society/sailing-robot #167: Try ROS Kinetic

Nanoseb, 2016-12-05:
We are currently using ros indigo on this project, but it is an old version, the latest one, kinetic have been released in june 2016. 

I have successfully installed it and ran our code with it (on my pi zero). I am now doing some more tests to see how better it is (if it is).

The installation of kinetic can be done along side with the current indigo one (just make sure to source the right setup.bash before any run and have a separate sailing-robot folder)

tsaoyu, 2017-04-04:
`catkin_make` was successful on ROS-kinetic with ubuntu 16.04
further function test with hardware need to be done in the future

tsaoyu, 2017-08-01:
All function tests passed on RPi3 with ROS kinetic. Now RPi3 + Kinetic is fully supported.

closed 2017-08-01 by tsaoyu