Maritime-Robotics-Student-Society/sailing-robot #171: Remove peak current draws from servo

smaria, 2016-12-08:
currently we can't use the same batteries for servo motor because during peak current draws the battery voltage collapses too much and the raspberry pi reboots. There must be a way to circumvent this with a capacitor based circuit, so we can simplify our power supply system.

alvaropp, 2016-12-08:
Sail servo : HS-785HB: current idle 8mA, running 230mA (both at 4.8V)
Rudder servo : S3003: current idle 25mA, running 100mA, stall 400mA (all at 4.8V)

alvaropp, 2016-12-08:
https://en.wikipedia.org/wiki/Battery_eliminator_circuit
http://www.stefanv.com/electronics/qf200105.html

smaria, 2016-12-08:
We should also try other power supplies, the issue we had might just be very specific to the one power bank we had

posecap, 2016-12-08:
would a 7.2v battery with separate regulator for the motors and electronics work?

posecap, 2016-12-08:
[image]
L78S05CV does 5v at 2amp then use a switching reg for the electric circuit