Maritime-Robotics-Student-Society/sailing-robot #172: Improve the wind sensor

alvaropp, 2016-12-08:
- Mechanical connection of new wind sensor
- 3D printed base for the current sensor

smaria, 2017-02-02:
I got some information how the tiny wind sensors are working that other people were using at the competition. It uses a mechanical encoder, which leads me to the conclusion that our magnetometer solution is still better.

I just realised there are plastic ball bearings - that sounds like a great idea as a basis for a mechanical connector: There shouldn't be much load on them and they can't be corroded by salt water

smaria, 2017-02-10:
http://www.igus.co.uk/wpck/2499/igubal_sphaerische_Axiallager?C=GB&L=en
I think this is what we want.

smaria, 2017-02-11:
http://simplybearings.co.uk/shop/Bearings-Deep-Groove-Ball-Bearings/c3_11/index.html

smaria, 2017-02-14:
http://www.bulletweights.com/Products/FishingSinkers/BulletWeights-Magnetic.aspx
this seems to be magnetic, and might make a good weight to balance the wind sensor

smaria, 2017-02-14:
Or just some magnets: https://www.amazon.co.uk/d/Magnets/POWER-MAGNET-STORE-Neodymium-Countersunk/B00L1LSOOQ/ref=sr_1_4?ie=UTF8&qid=1487102053&sr=8-4&keywords=magnets+hole

Nanoseb, 2017-02-14:
I think simple magnets are better since we want to have a 'clean' magnetic field and be able to orientate it so that the IMU sees it the best. With bullet weights you can't really do that, the best fit can be 90° from the 'natural' one loosing all the advantage of its shape.

smaria, 2017-02-14:
http://www.ebay.co.uk/itm/like/222285644558?lpid=122&chn=ps&adgroupid=38904455432&rlsatarget=pla-278029170694&adtype=pla&poi=&googleloc=1007089&device=c&campaignid=738208994&crdt=0
These might make an easy, light, no tools required attachment for the wind sensor and when using two they should give enough friction to not struggle with the sensor moving from zero rotation by accident

smaria, 2017-02-14:
http://www.ebay.co.uk/itm/Rubber-Stoppers-Bungs-10mm-Solid-Red-Pack-of-2-Laboratory-for-Test-Tubes-Science-/111824071937
these could help hold the magnet in place

smaria, 2017-07-18:
https://simplybearings.co.uk/shop/p170654/6882RS-Budget-Rubber-Sealed-Deep-Groove-Ball-Bearing-8x16x5mm/product_info.html

I purchased a couple of these for prototypes

takluyver, 2017-09-10:
We have a new wind vane assembly now. Have we tested this with an IMU attached?

Nanoseb, 2017-10-10:
Closing, see #205

closed 2017-10-10 by Nanoseb

CamilV, 2018-03-13:
Zirconium oxide bearings: 608CE Bearings
https://www.amazon.co.uk/Bearings-Zirconium-Ceramics-Skateboard-8%C3%9722%C3%977mm/dp/B06XRWK479/ref=sr_1_1?ie=UTF8&qid=1520964596&sr=8-1&keywords=608CE+bearing