Maritime-Robotics-Student-Society/sailing-robot #175: Consistent clock unit for Raspi

takluyver, 2017-02-07:
@tsaoyu mentioned this today - a device with a separate small battery which keeps running a clock, so that when we boot the Raspi up, it can immediately get the correct time, without needing to run the `setclock` script and get a GPS fix. Consistent timestamps would be really useful for organising recorded data like rosbags.

- How big are they, in volume & weight? Can they fit inside the inner box with Brian?
- How much do they cost?
- What physical & software integration do we need for the Raspi to get the time from one?

tsaoyu, 2017-02-07:
The one I found [here](http://www.volumerate.com/product/ds3231-raspberry-pi-rtc-board-real-time-clock-module-for-arduino-black-844277258#specifications-detailinfo) has the specifications:

Dimensions:	0.55 in x 0.55 in x 0.55 in (1.4 cm x 1.4 cm x 1.4 cm)
Weight: 5g
Cost: £1.74
It has  I2C standard interface

takluyver, 2017-02-08:
Sounds promising. Here's a tutorial on integrating it with the Raspi so the clock is set on boot:

https://learn.adafruit.com/adding-a-real-time-clock-to-raspberry-pi/overview

tsaoyu, 2017-05-01:
Before we order a real time clock module, can we have a last go with [NavCLOCK](https://github.com/tsaoyu/ublox/blob/master/ublox_msgs/msg/NavCLOCK.msg) message with latest GPS driver?

takluyver, 2017-05-02:
We do already get time messages from the GPS once ROS is running, and we use these in the GPS log that we produce. I'm happy to look at using the NavCLOCK message type for that, but I'm not aware of any problems with that part of the system.

However, the GPS time is inconvenient for a few reasons:

1. We want to have the system time on the Raspi correct before starting ROS, because it creates files with timestamps in the name as it starts up. We have the `utilities/setclock` script to do this, but we need to remember to run it. We forgot this at our test last weekend, for instance.
2. It's only available when the GPS receiver is connected and can pick up satellites, so we can't use it for an indoor test, for instance.
3. The NMEA 0183 interface we're using only gives us a time-of-day (e.g. 14:01:32), so we need to enter the date manually. I've just looked and the GPS signal does actually carry date information (in a rather weird format) - I don't know if there's a way to access that.

An alternative to the RTC module would be for one or more of our laptops to run ntp servers which the raspi can find and synchronise time from.

takluyver, 2017-07-03:
We have worked out how to get the date from the GPS as well, which improves this situation.

I have also written a script `piaccess/time2pi.sh` which sends the date & time from a laptop to the pi. It's a hackish way of doing it which won't give any degree of accuracy, but for our purposes it's sufficient to have the time accurate to a few seconds.

I want to test the time2pi script at the meeting tomorrow, and if it's working I'll close this.

takluyver, 2017-07-04:
OK, `time2pi.sh` is working.

closed 2017-07-04 by takluyver