Maritime-Robotics-Student-Society/sailing-robot #176: Indexing tool for rosbags

takluyver, 2017-02-07:
I'd like to write a tool that can index rosbags, so I can quickly see details like how long each one was recorded for and which topics it contains.

Nanoseb, 2017-02-15:
That is really nice!
Small improvement: right now the oldest date is first when the reverse is more useful

takluyver, 2017-02-15:
Thanks! I've reversed the sort order so the newest ones appear at the top.

smaria, 2017-02-22:
https://github.com/ewerlopes/rosbag-sliding-windows-annotator
not a finished thing, but they seem to be trying to do a similar thing to us

takluyver, 2017-02-22:
There's some similarity, but I think it's different from what I'm trying to do. I can't quite get from the description what sort of thing that would be used for, but they seem to be talking about annotating data inside a rosbag using images also stored in the rosbag. This tool is to give us an overview of our different rosbag files.

takluyver, 2017-07-03:
This is now there and working nicely:
https://github.com/Maritime-Robotics-Student-Society/sailing-robot/blob/master/utilities/index_recorded_data.py

Result:

[image]

The remaining change that I want to make is to add jinja2 to our docker image, because currently I have to launch docker and install it each time I want to use it.

takluyver, 2017-07-04:
OK, now if you don't have ROS, but do have docker, you can run the indexing script in a docker container like this:

```

takluyver, 2017-07-04:
OK, now if you don't have ROS, but do have docker, you can run the indexing script in a docker container like this:

```
./utilities/index_recorded_data.py --docker
```

closed 2017-07-04 by takluyver