Maritime-Robotics-Student-Society/sailing-robot #180: Force jibe

Nanoseb, 2017-02-15:
Force the boat to jibe when a tack fails.

I did a first implementation of that, but it was not used because quite dumb and could lead to more issues than without (it just creates a wp behind the boat). 

The idea would be to make a 'real' jibe like it is done when the jibe_to_turn parameter is set to true.

Nanoseb, 2017-02-15:
@takluyver at the moment your task manager can't subscribe to topics right? 
Do you think this can be done easily?

Here to implement this I want to have a node that time the duration of the tack and if it is too long it publish something to say to your class to jibe instead. Do you think that is the good approach?

takluyver, 2017-02-16:
The task manager handles a sequence of tasks (e.g. go to waypoint, or linger at waypoint), each of which has ways to calculate:

1. What should I do now?
2. Is this task finished? (when it is, the task manager will go on to the next one)

The manager has a Python API to insert a temporary task, do that and resume the task it was doing before. This is what it does with waypoints published on the `temporary_wp` topic. We could make a task type like 'jibe', or 'head downwind for 5 seconds', and then have it listen to another topic to trigger inserting such a task. I think that's simpler than something based on the obstacle avoidance waypoints, which is a specific kind of task.

Nanoseb, 2017-02-17:
Oh yes ok, that'll be cleaner, thanks

Nanoseb, 2017-02-22:
it is in progress, see branch [force_jibe](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/tree/force_jibe)

takluyver, 2017-04-23:
The implementation seems to work, though we might increase the timeout a bit for low wind conditions.

We also discussed yesterday using the proposed sensor-fusion speed estimates to delay tacking while the boat gains sufficient speed to make a tack, and to make decisions about when to jibe.

closed 2017-04-23 by takluyver