Maritime-Robotics-Student-Society/sailing-robot #181: What clever stuff can we do with an eInk screen?

takluyver, 2017-02-18:
I have picked up a second hand [Kobo Touch N905C](https://en.wikipedia.org/wiki/Kobo_Touch) eReader, which is highly hackable, to use as a display in bright light.

What can we do with that? In increasing order of complexity:

1. Translate manuals, checklists etc. into a format it can read and load them so we can easily see them in the light.
2. Generate images in the odd bitmap format the screen uses, and use a program already on the system to display them (I have already hacked it to give me telnet access).
  * Info here: http://www.chauveau-central.net/pub/KoboTouch/
3. Run Python on it and use available code to update part of the screen on demand
  * Python+Kobo info: https://www.mobileread.com/forums/showthread.php?t=219173
  * Updating display info: https://www.mobileread.com/forums/showthread.php?t=234002
  * Example application written in Python: https://bitbucket.org/david_weese/kobo-weather-app

If anyone's really keen, someone has even [got Debian running on a Kobo](https://sites.google.com/site/gibekm/hardware/kobo/kobo-as-tablet) with a lightweight Linux desktop. I don't think that's likely to be useful, but it's cool that it's possible.

smaria, 2017-02-22:
Sounds cool, I'm not sure what is really possible, these are the concepts I get from the above data, ordered by likelyness/usefulness:

- Have a laptop running as a webserver for the kobo, translating rosdata into $interesting stuff, display via wifi
- Have the device connected via usb, same as above
- Have the device connected to the Pi on the boat via usb(?), displaying $stuff
- Have the device running debian and then ROS itself, displaying stuff

Especially the first sounds like a really cool plan

takluyver, 2017-02-22:
The connectivity story:

* The device has wifi, though it has a tendency to turn off the radio after a few minutes to save power. Some people online think that you can kill a specific process to stop that happening - I haven't tested.
* The USB interface acts by default as a storage device (I think I saw something about running a network interface on it, but that starts to sound tricky). Maybe we could write files to it and have something running on the ereader that scans for an updated file and displays it? I think the device has to unmount storage to expose it on USB, though.

takluyver, 2017-11-01:
Another link in the 'advanced' category: [kobowm](https://github.com/pasqu4le/kobowm) is a simple X window manager written in Python.