Maritime-Robotics-Student-Society/sailing-robot #185: Try AHRS to get stable heading

tsaoyu, 2017-07-03:
Attitude and heading reference system (AHRS) use raw sensor data from the magnetometor, accelerometer, and gyro to compute a more stable heading. 

Try AHRS system https://github.com/DavidEGrayson/minimu9-ahrs could help us improve the performance of heading estimation.

joshctaylor, 2017-07-22:
I highly recommend this AHRS algorithm for your application. It is based on a paper which prevents magnetometers from distorting the attitude estimate - something common with most implementations.

http://wiki.ros.org/imu_complementary_filter

tsaoyu, 2017-07-25:
Thank you for your suggestion. We have already implement this sensor filter node on our boat. But this haven't been included in our calibration routine and `imu/data` give full orientation need to be convert to our heading in current system.

tsaoyu, 2017-07-25:
Since `xsens_driver` for MTi3 sensor from Xsens and `complementary_filter` node both give full `imu/data` with full orientation, our first task is to use `imu/data` to get heading.

joshctaylor, 2017-07-26:
This makes sense, I use the xSens heading estimate for my experiments - it's accurate enough for my purposes but I find the orientation drift starts to become significant after 10 mins when comparing 7 identical (MTw) sensors. I imagine if they have recommended the MTi3 for your purpose it has different filter weighting more suitable for stable heading estimates. It's worth discussing your application with Monique Paulich from xSens as she is quite familiar with the algorithms they are using.

takluyver, 2017-09-10:
Closing this for now as it's superseded by using the Xsens. We can use other issues (such as #197) for working out difficulties with the Xsens.

closed 2017-09-10 by takluyver