Maritime-Robotics-Student-Society/sailing-robot #191: Investigate Xsens filter scenarios

takluyver, 2017-08-06:
We think we might be using the scenario which ignores the magnetic field.

- [ ] Figure out which scenario it is using.
- [ ] Decide which one we want to be the default (Low_mag_dep?)
- [ ] If necessary, modify the ROS driver code to set the scenario when it starts.

takluyver, 2017-08-08:
We found it was using one called `north_reference`. We have set it to `general`. Waiting to test it.

takluyver, 2017-08-08:
Progress: it no longer uses a fixed heading when you start it up. The headings still weren't right, though. We'll try doing a new MFM next time.

takluyver, 2017-08-09:
@tsaoyu I realised last night that the bad readings we were getting in our final outdoor test might be partly due to the 180 degrees 'correction' we put into the sensor_service_imu code on Saturday. If you're testing at the weekend, try taking that out again.

tsaoyu, 2017-08-09:
Good point. I assume that is the downside of the hardcode approach. 
I will look at that 🔨  during weekend.

tsaoyu, 2017-08-15:
Progress: a consistent heading reading is acquired with `north_reference` is as profile, `NED` framework with 90 degrees offset. Drift in heading still happens may because the calibration or slight move during the start up.

takluyver, 2017-08-23:
We're still not reliably getting the right numbers from this. Do we:

1. Carry on fiddling with the settings of the Xsens
2. Go back to the settings which reliably produced a number relative to the starting heading, and accept that we can only start that node with the boat on land.
3. Revert to using the MinIMU for now

joshctaylor, 2017-08-24:
Did you try talking to Monica at xSens? She's been the most useful when I had sensor issues. I found the best was was to call them.

takluyver, 2017-08-24:
Thanks Josh! Tony, do you have the necessary contact information?

joshctaylor, 2017-08-24:
Monique Paulich <Monique.Paulich@xsens.com>
+31 88 97367 00

If you are using the university phones use the following access number
to allow you to dial internationally

0843 375 1020 (Planet Talk)

tsaoyu, 2017-08-25:
After bit research on the technical issue, I managed to make Xsens give us consistent heading reference now. As a reference I put some important points here

1. Do not use Active Heading Stabilisation (AHS) even it sounds like the thing we are looking. What it is actually doing is to use a stable magnetic field as a reference to correct gyro bias. Because of this, once it is actively no matter which profile is used, all heading reads out are not magnetic filed referenced any more. 

2. Rule of thumb for the calibration: at least 3 minutes will sufficiently large angle covered and held it **really** still. No Gaussian distributed normalised magnetic residual means we are in trouble, those calibrations should be discarded and should be use for the missions.

3. Orientation rest are set of  tool that help us to realign IMU inside the boat. It works like this: reset heading (means align x axis to north), inclination(align tangle plane), and alignment (x axis align with the centre of boat). The store option will remember those setting and write it to sensor. We don't need to touch because we currently hard code the rotation in `imu_service` node now.

4. Another dangerous thing we shouldn't click is a little compass with save icon. This is for in run calibration (ICC) function, which assume the full calibration is not possible and click such bottom will write a new magnetic field matrix to the sensor. 

5. Move the system during no rotation period will result in a wrong gyroscope bias calibration which leads to a slow but bounded drift on the heading. We need to hold the boat still for at least 6 seconds when launching the system. 

6. Position of battery has a huge influence on the magnetic field, firmly attach battery at exact place on top of RPi otherwise an unbounded drift on heading will happen.

tsaoyu, 2017-08-25:
Attach screen shot of a successful calibration. 
[image]

takluyver, 2017-08-26:
Nice! Was that calibration with the raspi and the battery pack present?

tsaoyu, 2017-08-26:
No, this one was done without the presence of battery pack.
but the one with battery pack looks pretty same with this.

closed 2017-09-08 by takluyver