Maritime-Robotics-Student-Society/sailing-robot #198: Figure out how to calibrate Xsens without connecting to a laptop

takluyver, 2017-09-10:
To calibrate the Xsens, we currently have to unplug it from the raspi and plug it in to a laptop using USB to run the magnetic field manager GUI. We've only got this program compiled for x86/x64, so we can't run it directly on the pi. However, it can run on a data file collected separately, rather than reading the Xsens live.

- [ ] Record the necessary data format using the raspi
- [ ] Write the generated calibration model to the Xsens from the raspi
- [ ] Automate moving the data between a raspi and a laptop (optional)

Another option would be to connect to the raspi and use its USB port remotely - I don't know if this is possible or how we would do it.

tsaoyu, 2017-09-12:
The "another" option seems pretty neat to me. I found something claim to do this https://virtualhere.com/home

takluyver, 2017-09-12:
Aha, and that appears to be based on a Linux feature called usbip. I think we can probably figure out how to use that directly, rather than using a proprietary wrapper.

Nanoseb, 2017-11-07:
this looks super straight forward: https://www.howtoforge.com/how-to-set-up-a-usb-over-ip-server-and-client-with-ubuntu-10.04
usbip is probably in the repo for ubuntu.

takluyver, 2017-11-14:
I've been trying to set this up, but run into problems with the fact that all the packages on the raspi are out of date - I can't install new packages compatible with the old ones on there. @tsaoyu can you remember what the problem was with updating before? Can I try updating everything and see what breaks?

Nanoseb, 2017-11-14:
just tried between 2 up to date archlinux, and I can't make it work, I have an error at the very end (during the attach). It isn't as straight forward as I thought ;)

Nanoseb, 2017-12-14:
Done, thx @takluyver  !
d68b23b

closed 2017-12-14 by Nanoseb