Maritime-Robotics-Student-Society/sailing-robot #205: Put electronics on new wind sensor

smaria, 2017-10-10:
We have a shiny new wind vane, and wire layout that can fit GPS, WiFi and wind sensing in the waterproof cable we have, so we now only need one single wire up the mast.

This requires drawing the final layout (after discussing it with me), some taking apart of WiFi antennas, soldering and waterproofing.

This may also solve #139 , since the WiFi antenna will be higher above the water.

CamilV, 2017-10-31:
Placing the IMU at the highest possible location on the under the wind sensor is the best place to set it up because of the physical limitation of the structure of the wind sensor and at the same time it gives us the best resolution of it, without saturation. Maximum and minimum reading values from the magnetometer shown as below. 

|   | Min    | Max   |
|---|--------|-------|
| X | -12704 | -2279 |
| Y | -22216 | 20985 |
| Z | -17724 | 24201 |

CamilV, 2017-10-31:
- [ ] Water proof of IMU
- [ ] Water proof of GPS
- [ ] Integration test of new wind sensor

Nanoseb, 2018-04-29:
I think we can close this now

closed 2018-04-29 by Nanoseb