Maritime-Robotics-Student-Society/sailing-robot #211: Create a structure for post high level commands

Nanoseb, 2017-11-07:
The title is probably not very clear...
For now, the high level package gives order directly to the low level like "tack_to_port" etc. and the low level catches these messages and set the rudder to +- 45 for example.
However sometimes there is some more computing/thinking between the highlevel and the low level. Like sheeting out when we jibe, jibing instead of tacking, jibing is a tack fails. And in the future waiting for the right moment to tack, sheet out to tack, sheet out if strong heeling etc.
All of these things are not supposed to be handled by the very high level (that cares more about the path planning).
At the moment things that are implemented are a bit "hacky". Having a proper structure for them would be cleaner and easier to add new behaviour in the future.

I don't know yet how to handle that, one large node? one node per behaviour? something else?

Nanoseb, 2018-02-15:
@takluyver and all, I think we can continue the discussion we had on Tuesday here.

After thinking about it, I agree that the voting system should be in this manoeuvre/helming node, it is not sailing strategy/navigation but  something to prevent from tacking too much. However the highlevel should still have a way to overwrite this (tack right now) when close enough to the waypoint etc (needed for the station keeping for ex). But this can be implemented later, for now we can keep the voting in the highlevel.

Then, concerning the structure of such code, the more I think about it and the more I think it will look like our task system... ROS nodes are not made for that, you can't really run a node for a short amount of time and kill it afterwards etc.
So what I see is:
   - Helming node subscribes to `sailing-state` and `goal heading`
   - we have one class per behaviour (tack, jibe, tack-in-wave, tack-with-sheet-out, tack-with-speed-up-phase etc.)
   - these classes need some parameters like which direction is the tack/jibe
   - the timeout check can be left to the Helming node, it can stop a behaviour class from running and go to the next one on its list if needed. Or come back to the normal behaviour if the tack is a success.

One thing I still don't know is how to handle the sheet sail setting. The behaviour class sometimes need to overwrite these (sail setting not based on wind direction any more), but sometime they don't. And having the same code in each and every behaviour class to handle sailsettings isn't nice. Same goes for the rudder, sometimes in the behaviour code we will need to keep using the PID and then overwrite to +-45 degree. Not sure yet how to do that cleanly.

takluyver, 2018-02-15:
I think that mostly sounds good, but one minor note: I think having the high level node override the lower level is probably bad design. I don't have a neat rule or design pattern to point to for this, but my instinct is that it would be messy. We'd need some extra code to decide when to listen to the high level instead of the low level, and then we'll find some case where we want the lower level to override the higher level again, and pretty quickly you've got a digital argument which is a pain to debug.

So I'd try to design it without overrides as far as possible. For each thing the system has to decide, there should be one node which makes that decision. For other nodes, let's think in terms of sending information that can affect the decision, rather than overriding it.

Nanoseb, 2018-02-15:
Yes I totally agree with you for the overwriting. I said it in those terms because I didn't really know how to make these things work.

Something we could do, I think, it is to have some functions that act on the low level and that are easy to use like for example for the rudder:
- set_rudder(PID, goal_heading)
- set_rudder(-5degree) 
- set_rudder(MAX_ANGLE) for example

same for the sail: 
- set_sail(wind)
- set_sail(OUT)
- set_sail(IN)
- set_sail(wind, 1.2) (=to sheet out 20%more than the lookup table for example)

Then it will make the behaviour class super easy to write, just focusing on the logic

Nanoseb, 2018-02-15:
Ah yes, sorry you were talking about the tack now thing vs the voting system. Yes ok, that makes sense, then for example in this case the helming node will need to know the distance to the waypoint, but that's ok, it can be published by the high level.

I am going to try to draw a schematic of all the information passing between nodes/components

Nanoseb, 2018-04-17:
2003652 is a first working version, it still needs debugging (with the simulator), cleaning, and re-factoring.
I think this file can (and should) be split in 2 or 3 files.

Nanoseb, 2018-04-26:
see #237

Nanoseb, 2018-04-27:
This is now done and merged, it will be tested on water soon. Closing this and I'll reopen more specific issues if needed.

closed 2018-04-27 by Nanoseb