Maritime-Robotics-Student-Society/sailing-robot #227: Generate .KMZ/.KML gps trace

Nanoseb, 2018-02-07:
A .kmz can be read by almost any maps like openstreet map/google earth/map etc. It could be a nice addition to the index_recorded_data.py script to have an osm map summarizing the bagfile.

Such file can be easily generated from the .csv we already record with pyKML for example (no need to parse the huge bagfile or generate this live).

Nanoseb, 2018-02-07:
http://osm.quelltextlich.at/wizard-kml-osmtools.html for example to generate easily the map afterwards

takluyver, 2018-02-07:
The [folium](http://folium.readthedocs.io/en/latest/) library can plot over Openstreetmap directly, without needing to make a kml file. I used this last year to plot the tracks from the competition:

http://nbviewer.jupyter.org/github/Maritime-Robotics-Student-Society/sailing-robot/blob/master/notes/WRSC%202017%20results.ipynb

Nanoseb, 2018-02-07:
Oh yes nice, I didn't realised you got an actual osm thing. So it may be even easier to add to the index page. 
Definitely worth looking at then.

Nanoseb, 2018-02-08:
Basic version working \o/, still needs waypoints now
[image]

takluyver, 2018-02-08:
Nice! It should be easy to get the waypoints from the parameter dump.

Nanoseb, 2018-02-08:
yes, it is just not a copy/paste from your code since it is json vs yaml ;). But yes shouldn't be complicated

Nanoseb, 2018-02-08:
@takluyver can you test with all the bags/.csv? 
If there is no loading issue and it works fine I think we can close this.

takluyver, 2018-02-10:
We are going to need to be smarter with the loading, unfortunately. When I generated the index with all the maps, it took over a minute to load in my browser and made 1500 requests. I'll see if I can figure something out.

takluyver, 2018-02-10:
OK, now it only loads the map when you click the link to open it for the first time.

closed 2018-02-10 by takluyver