Maritime-Robotics-Student-Society/sailing-robot #230: uBlox GPS with i2c

Nanoseb, 2018-03-17:
So, today we (Camil and I) have played a bit more with the GPS and i2c.
A couple of things can be noted. First i2c seems to be smart enough and there is no way of having clash of data with imu's for example (it was fun to see that with the logic analyser: imu message in the middle of the GPS ones ;) )

Secondly, we have about 40% of the gps messages that do not pass the checksum. It is only one or two characters that are not read correctly but that makes the message harder to parse so what to do with that:
* is it an issue? can we just assume that the GPS will work at half of its frequency and still be fine
* since only a couple of the characters are not good can we write a parser that will be more robust, the probability that the actual GPS coordinates are not readable is even lower

A last thing to consider is that with i2c we can't reprogram the chip, this means that we probably need to have the 2 pins accessible in case we need that. 

The code we used to test the GPS is there: https://github.com/Maritime-Robotics-Student-Society/sailing-robot/blob/master/utilities/test_i2c_gps.py

@tsaoyu or @takluyver maybe? an idea?

If i2c isn't good enough we can still use the serial for the GPS, but we are avoiding that because it adds 2 cables to the top of the mast.

takluyver, 2018-03-18:
If it's purely random, it's probably fine to skip the corrupted messages and accept a lower frequency. If there are blocks of several seconds where all the messages are corrupted, that might be more of a problem, depending on what the boat is doing at the time.

I'd say it's a bad idea to make the parser more 'robust' and try to use corrupted data: sooner or later, the corruption will give you a valid coordinate that's hundreds of miles out.

Was the 40% error rate with a long connection or a short one? Does it increase with the wire length? Are there things we can do (e.g. twisting wires together) that might improve the reliability? I've got no idea what sort of signals i2c uses.

Nanoseb, 2018-03-18:
Yes, it looks random, but an other thing that will be interesting to analyse is the actual rate that we can achieve.

Also we will need to try outside with actual GPS data, here messages were quite empty, with longer messages maybe the probability of them being corrupted is higher.

The 40% were gathered with long wires, and they were twisted even if in this case it shouldn't have an impact on the signal.

Concerning having a more "robust" parser, ok yes it is probably a bit dangerous. Also the checksum is just a XOR on the ascii character sent (which is quite weak I think), this means that some of the message have the right checksum but are still corrupted (even number of bit flipped at the right position for example).

CamilV, 2018-03-18:
I don't think the I2C signal is an issue in this case as the IMu, which is running on the same lines, doesn't lose any data, or at least we didn't notice it losing data. The signal is a bit noisy, because of the long wires and the close proximity of them, but once converted into digital, it doesn't seem to be an issue.
[image] 
Channel 0 is SCL and channel 1 is SDA.
You can see the noisy signal here, when one line changes state, it pulls the other line with it by about 0.25V, but I don't think that's an issue, as the digital conversion looks alright and I2C only uses the 2 voltage levels, low and high. We could try having the 2 lines further apart, in that case we wouldn't have this noise, but I'm pretty certain that won't have any influence on the corrupted bits

Nanoseb, 2018-04-11:
We've made some more test yesterday, and with i2c we can achieve only a frequency of about 0.4-0.5 Hz... Which is not enough. Also the settings set with the weird ublox messages (done with serial) doesn't seems to survive a reboot of the gps. This would be usefull to improve the frequency rate of the gps.

CamilV, 2018-04-17:
https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_(UBX-13003221)_Public.pdf

Nanoseb, 2018-04-17:
So today some quite good news:
- we managed to send ublox messages to the gps via i2c
- we have a script in utility (for now) to send less useless messages and increase the frequency to 5Hz
- result: we get about 3.7-4Hz in the end \0/

So we are quite happy with this and I think we can continue with i2c, just have to write the node to use that now (or adapt the current node for i2c).

Nanoseb, 2018-04-27:
I think we have this issue: http://www.advamation.com/knowhow/raspberrypi/rpi-i2c-bug.html

Nanoseb, 2018-04-27:
Ok, yes it seems to be that, it leads to i2c hanging when the gps and the pi are out of sync, and only a reboot can help... 
To avoid this we can reduce the i2c baud rate (line in /boot/config.conf) but this now leads to timeout when sending/receiving messages. So we need to figure out how to increase the i2c timeout.
It seems doable, but this: https://raspihats.com/i2c-hat/2016/02/16/raspberry-pi-i2c-clock-stretch-timeout.html doesn't work apparently (probably out of date).

Nanoseb, 2018-05-15:
we tried it, and it works fine!!!

closed 2018-05-15 by Nanoseb