Maritime-Robotics-Student-Society/sailing-robot #233: Fix docker repository

Nanoseb, 2018-04-18:
When trying to run `docker pull jamak9/sailing-robot` I get: 
```
Error response from daemon: pull access denied for jamak9/sailing-robot, repository does not exist or may require 'docker login'
```
What can be done about this?

takluyver, 2018-04-18:
All references to `jamak9/sailing-robot` should now be replaced with `sotonsailbot/ros:indigo` . If you use the scripts in `utilities/docker/` they should already be using the new name.

Nanoseb, 2018-04-18:
ok, I'll update the wiki then, thx

takluyver, 2018-04-18:
PS the images are stored in the [sotonsailrobot organisation](https://hub.docker.com/r/sotonsailbot/) on Docker Hub. Currently I'm the only member of the organisation, but I can add anyone else. You need to be a member to publish a new docker image to that tag (though you can always change the dockerfiles and build the image directly, with no need to push to Docker Hub).

closed 2018-04-18 by Nanoseb