Maritime-Robotics-Student-Society/sailing-robot #234: Installation: testresources, tornado, setup.py

bryzhao, 2018-04-21:
Hi Soton team,

During the installation step on the sailing-robot wiki (https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Virtual-Machine-for-Windows-Mac-users), under the **Post installation setup** section, where the instructions say:

- Install the needed dependencies for the sailing robot

`~/sailing-robot/utilities/setup_scripts/Install_needed_packages.sh`

I get the following errors in my terminal (in bold):
Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version.
gpsd-clients is already the newest version.
i2c-tools is already the newest version.
libgeos-dev is already the newest version.
python-scipy is already the newest version.
vim is already the newest version.
python-pip is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 296 not upgraded.
Installing Python packages...
The directory '/home/bryanz/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/bryanz/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
/usr/local/lib/python2.7/dist-packages/pip/_vendor/urllib3/util/ssl_.py:339: SNIMissingWarning: An HTTPS request has been made, but the SNI (Subject Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  SNIMissingWarning
/usr/local/lib/python2.7/dist-packages/pip/_vendor/urllib3/util/ssl_.py:137: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
 InsecurePlatformWarning
Requirement already up-to-date: pip in /usr/local/lib/python2.7/dist-packages (10.0.1)
**launchpadlib 1.10.2 requires testresources, which is not installed.
matplotlib 1.3.1 requires tornado, which is not installed.**
The directory '/home/bryanz/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/bryanz/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting Latlon
/usr/local/lib/python2.7/dist-packages/pip/_vendor/urllib3/util/ssl_.py:339: SNIMissingWarning: An HTTPS request has been made, but the SNI (Subject Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  SNIMissingWarning
/usr/local/lib/python2.7/dist-packages/pip/_vendor/urllib3/util/ssl_.py:137: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  InsecurePlatformWarning
/usr/local/lib/python2.7/dist-packages/pip/_vendor/urllib3/util/ssl_.py:137: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  InsecurePlatformWarning
  Downloading https://files.pythonhosted.org/packages/cf/f0/5c23d9431a69cf27d139e9ae09db7f0f032c88192384dad3f8d7e0c860e8/LatLon-1.0.2-py2-none-any.whl
Collecting shapely
  Downloading https://files.pythonhosted.org/packages/a5/cf/0d51a15861d944da8463d8c1aecf39cfc6852ea9ecd092046824e9b237b3/Shapely-1.6.4.post1-cp27-cp27mu-manylinux1_x86_64.whl (1.5MB)
    100% |████████████████████████████████| 1.5MB 6.7MB/s 
Collecting pyproj
  Downloading https://files.pythonhosted.org/packages/29/72/5c1888c4948a0c7b736d10e0f0f69966e7c0874a660222ed0a2c2c6daa9f/pyproj-1.9.5.1.tar.gz (4.4MB)
    100% |████████████████████████████████| 4.4MB 1.6MB/s 
Collecting pynmea2
  Downloading https://files.pythonhosted.org/packages/7c/82/f734e75dbb0aecbd9ea20489f364fdac9f124f17c4d4e26d2ab160c8ff58/pynmea2-1.12.0.tar.gz
Collecting spidev
  Downloading https://files.pythonhosted.org/packages/36/83/73748b6e1819b57d8e1df8090200195cdae33aaa22a49a91ded16785eedd/spidev-3.2.tar.gz
Collecting tornado
  Downloading https://files.pythonhosted.org/packages/cf/d1/3be271ae5eba9fb59df63c9891fdc7d8044b999e8ac145994cdbfd2ae66a/tornado-5.0.2.tar.gz (506kB)
    100% |████████████████████████████████| 512kB 9.5MB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-IuomJA/tornado/setup.py", line 146, in <module>
        raise ImportError("Tornado requires an up-to-date SSL module. This means "
    ImportError: Tornado requires an up-to-date SSL module. This means Python 2.7.9+ or 3.4+ (although some distributions have backported the necessary changes to older versions).
    
    ----------------------------------------
**Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-IuomJA/tornado/**

So far, I have tried `pip install --user launchpadlib==1.10.2`, as well as updating to Python 3.6.x, and then re-running the installation instruction, but I still get the same errors.

Any help would be much appreciated. Thank you!

Nanoseb, 2018-04-23:
I have seen you have tried the docker image too, I would recommend using that instead of virtual box, it is easier to set up, specially to have a share folder between your computer and the virtual environment.

bryzhao, 2018-04-24:
@takluyver @Nanoseb thanks for the reply. I used `pip install launchpadlib == 1.10.2 --user` to install testresources, and `pip install tornado==4.1 --user` to install a tornado version (older than 5). Changing the bash script to `pip install "tornado<5"` unfortunately did not work for me. 

Additionally, I encountered two more errors after running the installation script - in which the packages pigpio and python-pigpio are not located (in bold at the bottom):

Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version.
gpsd-clients is already the newest version.
i2c-tools is already the newest version.
libgeos-dev is already the newest version.
python-scipy is already the newest version.
vim is already the newest version.
python-pip is already the newest version.
The following packages were automatically installed and are no longer required:
  cmake cmake-data collada-dom-dev collada-dom2.4-dp-base
  collada-dom2.4-dp-dev freeglut3 gir1.2-gtk-2.0 hddtemp icu-devtools
  liballegro4.4 liballegro4.4-plugin-alsa libapr1-dev libaprutil1-dev
  libassimp-dev libassimp3 libatk1.0-dev libavcodec-dev libavformat-dev
  libavutil-dev libboost-all-dev libboost-atomic-dev libboost-atomic1.54-dev
  libboost-atomic1.54.0 libboost-chrono-dev libboost-chrono1.54-dev
  libboost-chrono1.54.0 libboost-context-dev libboost-context1.54-dev
  libboost-context1.54.0 libboost-coroutine-dev libboost-coroutine1.54-dev
  libboost-date-time-dev libboost-date-time1.54-dev libboost-dev
  libboost-exception-dev libboost-exception1.54-dev libboost-filesystem-dev
  libboost-filesystem1.54-dev libboost-filesystem1.54.0 libboost-graph-dev
  libboost-graph-parallel-dev libboost-graph-parallel1.54-dev
  libboost-graph-parallel1.54.0 libboost-graph1.54-dev libboost-graph1.54.0
  libboost-iostreams-dev libboost-iostreams1.54-dev libboost-iostreams1.54.0
  libboost-locale-dev libboost-locale1.54-dev libboost-locale1.54.0
  libboost-log-dev libboost-log1.54-dev libboost-log1.54.0 libboost-math-dev
  libboost-math1.54-dev libboost-math1.54.0 libboost-mpi-dev
  libboost-mpi-python-dev libboost-mpi-python1.54-dev
  libboost-mpi-python1.54.0 libboost-mpi1.54-dev libboost-mpi1.54.0
  libboost-program-options-dev libboost-program-options1.54-dev
  libboost-program-options1.54.0 libboost-python-dev libboost-python1.54-dev
  libboost-python1.54.0 libboost-random-dev libboost-random1.54-dev
  libboost-random1.54.0 libboost-regex-dev libboost-regex1.54-dev
  libboost-regex1.54.0 libboost-serialization-dev
  libboost-serialization1.54-dev libboost-serialization1.54.0
  libboost-signals-dev libboost-signals1.54-dev libboost-signals1.54.0
  libboost-system-dev libboost-system1.54-dev libboost-test-dev
  libboost-test1.54-dev libboost-test1.54.0 libboost-thread-dev
  libboost-thread1.54-dev libboost-thread1.54.0 libboost-timer-dev
  libboost-timer1.54-dev libboost-timer1.54.0 libboost-tools-dev
  libboost-wave-dev libboost-wave1.54-dev libboost-wave1.54.0 libboost1.54-dev
  libboost1.54-tools-dev libcairo-script-interpreter2 libcairo2-dev
  libcegui-mk2-0.7.6 libcf0 libconsole-bridge-dev libconsole-bridge0.2 libcr0
  libcurl4-openssl-dev libcv-dev libcvaux-dev libdc1394-22-dev libdevil-dev
  libdevil1c2 libeigen3-dev libflann-dev libflann1.8 libfltk1.1 libfltk1.1-dev
  libfreeimage-dev libfreeimage3 libgcrypt11-dev libgdk-pixbuf2.0-dev
  libgl2ps-dev libgl2ps0 libglib2.0-dev libgnomecanvas2-0
  libgnomecanvas2-common libgnutls-dev libgnutlsxx27 libgpg-error-dev
  libgsl0ldbl libgtest-dev libgtk2.0-dev libharfbuzz-dev libharfbuzz-gobject0
  libhdf5-7 libhighgui-dev libhwloc-dev libhwloc-plugins libhwloc5
  libibverbs-dev libibverbs1 libicu-dev libidn11-dev libilmbase-dev
  libjasper-dev libjbig-dev libjpeg-dev libjpeg-turbo8-dev libjpeg8-dev
  libkrb5-dev liblcms1 liblcms1-dev liblcms2-dev libldap2-dev liblodo3.0
  liblog4cxx10 liblog4cxx10-dev liblua5.1-0 liblz4-1 liblz4-dev liblzma-dev
  libmng-dev libmng2 libnetcdf-dev libnetcdfc++4 libnetcdfc7 libnetcdff5
  libodbc1 libogg-dev libogre-1.8-dev libogre-1.8.0 libois-1.3.0
  libopencv-calib3d-dev libopencv-contrib-dev libopencv-core-dev libopencv-dev
  libopencv-features2d-dev libopencv-flann-dev libopencv-gpu-dev
  libopencv-gpu2.4 libopencv-highgui-dev libopencv-imgproc-dev
  libopencv-legacy-dev libopencv-ml-dev libopencv-objdetect-dev
  libopencv-ocl-dev libopencv-ocl2.4 libopencv-photo-dev libopencv-photo2.4
  libopencv-stitching-dev libopencv-stitching2.4 libopencv-superres-dev
  libopencv-superres2.4 libopencv-ts-dev libopencv-ts2.4 libopencv-video-dev
  libopencv-videostab-dev libopencv-videostab2.4 libopencv2.4-java
  libopencv2.4-jni libopenexr-dev libopenmpi-dev libopenmpi1.6 libopenni-dev
[... 143 lines cut ...]
Requirement already satisfied: pyproj in /usr/local/lib/python2.7/dist-packages (1.9.5.1)
Requirement already satisfied: pynmea2 in /usr/local/lib/python2.7/dist-packages (1.12.0)
Requirement already satisfied: spidev in /usr/local/lib/python2.7/dist-packages (3.2)
Requirement already satisfied: tornado in ./.local/lib/python2.7/site-packages (4.1)
Requirement already satisfied: pi_ina219 in /usr/local/lib/python2.7/dist-packages (1.1.0)
Requirement already satisfied: certifi in ./.local/lib/python2.7/site-packages (from tornado) (2018.4.16)
Requirement already satisfied: backports.ssl-match-hostname in ./.local/lib/python2.7/site-packages (from tornado) (3.5.0.1)
[... cut ...]
Setting timezone...
Installing pigpio from apt...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
**E: Unable to locate package pigpio
E: Unable to locate package python-pigpio**

Also, @Nanoseb, could you elaborate on what you mean for using the docker image instead of the virtual box? Currently, I have tried launching the ROS simulator nodes from the docker image, which I access within my virtual machine, but the ROS nodes fail. 

Any ideas would be appreciated :)

takluyver, 2018-04-24:
Looking at that setup script, I think it's maybe only meant to run on the raspberry pi - it installs a number of packages for interfacing with hardware (like pigpio, pi_ina219, i2c_tools, spidev...). We should make that clearer in the naming and/or description.

The whole setup scripts folder probably needs tidying up - there are several scripts that haven't been touched in a couple of years and are probably outdated. And it's not clear what is meant to run where, or in what order.

What failures do you see when you try to launch the simulator in the docker container? Sorry if you already posted this elsewhere - I get a lot of emails and github notifications. :-)

tsaoyu, 2018-04-24:
@takluyver are you mention [this](https://downloads.ubiquityrobotics.com/pi.html)?

takluyver, 2018-04-24:
Thanks, that was what I was thinking of. It looks like they also have a VM image, but there are no obvious details on what versions of things it has. https://downloads.ubiquityrobotics.com/vm.html

closed 2018-09-02 by Nanoseb