Maritime-Robotics-Student-Society/sailing-robot #235: Update Docker and Vm images

Nanoseb, 2018-04-23:
The indigo-ros-base docker base image and the VM image ([here](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Virtual-Machine-for-Windows-Mac-users)) are a bit old.

2 things need updating:

- switch to kinetic (instead of indigo, we can maybe even try lunar, it is what I have installed on my laptop and the code works fine with it (but we should check first if it is available for the pi)).

- use a more recent distribution, ubuntu 18.04 is also a LTS and will come with newer version of packages (like python...) (see #234).

takluyver, 2018-04-23:
18.04 isn't even out yet! And we should probably give it a few weeks at
least once it is, to let the dust settle.

Updating Ros makes sense, though. I think we're already using kinetic on
the pi.

Nanoseb, 2018-04-23:
Yes for ubuntu let's wait a bit ;) (I am too used to arch I guess)

And yes, we have kinetic on the pi.