Maritime-Robotics-Student-Society/sailing-robot #236: Update Wiki documentation

Nanoseb, 2018-04-23:
I went through the wiki here is the list of pages that needs updating:

- [ ] [Naming convention](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Naming-conventions)
- [ ] [Install ROS](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Install-of-ROS): install Kinetic instead, or just redirect to ROS documentation
- [ ] [Using the repository](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Using-the-repository): talk about post install scripts
- [ ] [2D plot of the boat](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Rviz:-2D-plot-of-our-boat): Add the new 2D plot thing using matplot
- [ ] [Rosbag analysis](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/ROS-bag-analysis): add more things there
- [ ] [Electronic-hardware-list](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Electronic-hardware-list): update the list with new GPS and IMU (Xsens)
- [ ] [Schematic-diagram](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Schematic-diagram): update with GPS using i2c and usb on top of the mast
- [ ] [Naming convention](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Naming-conventions): needs a huge update I think, no one is using this page anymore because it is not uptodate (maybe replace it by a automatically generated version)
- [ ] [Pre sail Raspi setup](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Pre-sail-Raspi-setup): seems quite old, probably needs updating
- [ ] [Push to pi](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Git-Push-on-Raspberry-(offline-Git)): use the script in piaccess instead
- [ ] [Utililty scripts](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Utility-scripts): add an intro, this page is a bit unclear I think, and add other scripts from the utility folder
- [ ] [Itchen-Lake-Water-Test](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Itchen-Lake-Water-Test): We have lost the image?

PS: To those using the WIKI (@bryzhao ?), if a page is listed here it is not fully up to date ;)

bryzhao, 2018-04-24:
Thank you @Nanoseb!

Nanoseb, 2018-05-17:
This is **really** good! thanks a lot!
A couple of things could maybe be improved, but it is already very good:
- mention that M3/CH3 is the rudder and M1/CH1 the sail servo
- Say that the white rectangle on the right is a usb connector
- name pins 1 and 8 like the other in the waterproof connector
- repeat M1/M3/FEEDBACK/GND on the right side of the waterproof connector
- name the pins on the 4 pin wp connector (but I guess you need to have the actual part in front of you for that)
- Mention the voltage of the battery pack maybe?
- Above each picture give the component/part number
- Maybe name the pins on the IMU and on the multiplexer, I think it would help understanding what they do

Thanks again for this (and if you don't have time right now to do these modifications that's all right, it can wait a couple of weeks)

Nanoseb, 2018-05-25:
I've updated the wiki with the wiring, I had to export a .png though, because svg on wiki doesn't accept images embedded in the svg...