Maritime-Robotics-Student-Society/sailing-robot #238: Using the repository - launching nodes

bryzhao, 2018-04-28:
Hi Soton team,

Hope you are all doing well.

We have now installed ROS Kinetic on our Pi (3), and I have tried following the steps from the documentation up until the "launching nodes" step under the "Using repository" section. 

The post-installation setup step to install the dependencies (~/sailing-robot/utilities/setup_scripts/Install_needed_packages.sh) only yielded 1 error this time, for the missing "testresources" package, so I simply used `pip install launchpadlib==1.10.2 --user` and solved the problem that way. The installation script worked fine after that.

All of the other steps worked nicely up until I tried running: `roslaunch sailing_robot test-all-systems.launch`, under the [Using repository](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Using-the-repository) step. I got these following errors:

auto-starting new master
process[master]: started with pid [32289]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 90088a72-4a86-11e8-b01f-b827eb67a500
process[rosout-1]: started with pid [32302]
started core service [/rosout]
process[dummy_apparent_wind-2]: started with pid [32319]
process[dummy_heading-3]: started with pid [32320]
process[dummy_position-4]: started with pid [32321]
**ERROR: cannot launch node of type [sailing_robot/heading_control]: can't locate node [heading_control] in package [sailing_robot]**
process[tack-6]: started with pid [32322]
process[actuator_demand_rudder-7]: started with pid [32323]
**ERROR: cannot launch node of type [rosserial_python/serial_node.py]: rosserial_python
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/sailboat/sailing-robot/src
ROS path [2]=/opt/ros/kinetic/share**

I was wondering – what exactly does launching the nodes do? Am I able to launch the nodes even if there is no hardware attached to the Pi?
We have our own hardware design (using an Arduino Mega, anemometer, RPi, GPS, XBee modules), but are wondering if the code provided by the Southampton team is adaptable or easily changed to fit another set of similar hardware components.

Thank you all!

takluyver, 2018-04-28:
I would start with `simulation.launch` rather than `test-all-systems.launch`. You can see the different [launch files in this folder](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/tree/master/src/sailing_robot/launch). Any that are under 'archive' (like `test-all-systems.launch`) may well not work any more. It's not guaranteed that the others work either, but there's a better chance.

Inside a launch file, there are a bunch of lines like this:

```xml
<node name = "simulation_position" pkg='sailing_robot' type="simulation_position"/>
```

The `type=` part will generally correspond to the name of a file in [the scripts folder](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/tree/master/src/sailing_robot/scripts).

If we've got things right, only the nodes called `sensor_driver_*` and `actuator_driver_*` talk to the hardware. Where your hardware differs from ours, you may need to write/find different 'driver' nodes for that. The other nodes, which do all the calculation and decision making, work by sending and receiving ROS messages. A sensor driver node (like `sensor_driver_gps`) gets a reading from the sensor and sends out one or more ROS messages to other nodes. At the other end, an actuator driver node listens for relevant ROS messages and uses them to update the servo PWM.

The simulation launch file doesn't use any driver nodes, so it's a good way to test that you've got ROS and the control nodes working before you start trying to integrate it with hardware.

Nanoseb, 2018-09-02:
Closing this now, I think it is clarified.

closed 2018-09-02 by Nanoseb