Maritime-Robotics-Student-Society/sailing-robot #239: usbip setup failure

Nanoseb, 2018-04-29:
So today we tried to use the `usbip-bind.sh` script, and... it fails ;)

Here is the error we got:
```
SSH-ing to Pi to set up usbip export...
pi
Bus ID: 1-1.5
[sudo] password for pi: 
usbip: info: bind device on busid 1-1.5: complete
Connection to 192.168.12.1 closed.
Attaching [-r 192.168.12.1 -b 1-1.5]
libusbip: error: udev_device_new_from_subsystem_sysname failed
usbip: error: open vhci_driver
usbip: error: query
```
We will try to investigate that during next meeting, @takluyver maybe you have an idea of the source?

takluyver, 2018-04-29:
There's some kernel module that you need loaded on Tabarly (or whatever computer you're trying to use). Try `sudo modprobe vhci-hcd` first. If that's not it, try tab completing after `vhci`, because I think it started with that.

takluyver, 2018-04-29:
I hope that's it, I'm not 100% sure. I think there are different modules that need to be loaded on each end of the connection.

Nanoseb, 2018-04-29:
ok, we will try on Tuesday, we went for the cable link today after trying a couple of times. Probably worth loading these modules in the script in case they are not loaded beforehand so that it is easier to operate.

tsaoyu, 2018-04-30:
Sure, we can try this on Tuesday.

commit b0c22430 (2018-05-01): fix #239
closed 2018-05-01 by tsaoyu