Maritime-Robotics-Student-Society/sailing-robot #242: Multiplexer Wiring

jordanjj8, 2018-05-10:
Hi Southampton Team,

We are going to take on your guy's advice and use a multiplexer for our system :)

We were just wondering, how did you guys wire up and use the multiplexer? (specifically, how did you wire up a switch on the transmitter and for it to switch modes on the multiplexer side?)

Possibly a wiring diagram would help us very much! Thank you again and hope to hear from you soon!

Best,
UC Davis Sailing Robotics Team

Nanoseb, 2018-05-10:
Hi,
Is this page clear enough? https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Schematic-diagram#for-the-multiplexer

It should have all the information needed, the selector is  wired on channel 5 of our remote control.
M1-4 are "master" so the raspberry pi in our case and S1-4 are the slave input (the remote control). This is done that way in order to have the master controlling the servos when the remote is out of range by the boat.
Servos are connected to OUT1-4 (1 and 3 in our case)

Nanoseb, 2018-05-10:
Oh sorry I didn't read your question carefully enough, we don't wire anything on the remote, our remote have a large number of channels (I would say 8), we have 4 switches we can use for that on our remote. The switch on the remote changes the value of channel 5 on the receiver that is read by the multiplexer.

jordanjj8, 2018-05-11:
Thank you for the post. Much appreciated.

What was your channel 5 (channel that controls the SEL) ? Was it a simple switch? 

Also, did you have to do any multiplexer calibrating?

Nanoseb, 2018-05-31:
I think the question have been answered, I am closing this issue. Feel free to open a new one if needed.

closed 2018-05-31 by Nanoseb