Maritime-Robotics-Student-Society/sailing-robot #247: Investigate GPS jumps

Nanoseb, 2018-05-16:
This week end we've tried the small ublx gps (the one without the antena). The boat position jump of up to 20m from its previous position. See bag files from 13/05/2018. 

To investigate that, letting the GPS run while in a stationary position is probably the 1st thing to do.

takluyver, 2018-05-16:
Might also be interesting to check how many satellites it can see when that happens - presumably more satellites give a more precise fix.

Nanoseb, 2018-05-17:
We checked that, it does not seem to be related, the number of satellites is always above 10, and doesn't even change from 12 to 10 when it happens.

Nanoseb, 2018-09-01:
It worked quite well during the competition