Maritime-Robotics-Student-Society/sailing-robot #249: Add map background to 2D plot

Nanoseb, 2018-05-16:
We have a brand new 2D plot tool, it could be nice to have a map in the background.

Here is a couple of lib that could work:
* https://stackoverflow.com/questions/28476117/easy-openstreetmap-tile-displaying-for-python
* https://github.com/MatthewDaws/TileMapBase
* https://github.com/rossant/smopy

TAJD, 2018-05-17:
I don't know what sort of things you are using/looking at but I've had success with using [Bokeh](https://bokeh.pydata.org/en/latest/ ) which uses Google in the background.

The strong/weak point is the plotting functionality - it isn't quite as modifiable as pyplot but it still looks good. What is quite fun is the ability to run little web apps written basically in code file which allow you to play with the data and plot it real time.

tsaoyu, 2018-06-03:
We had a bit trouble on 2D plot when testing on this Sat. 
The problem is the plot region is hard coded and no GPS is shown when boat is far away from these waypoints: 

https://github.com/Maritime-Robotics-Student-Society/sailing-robot/blob/master/src/sailing_robot/scripts/debugging_2D_plot_matplot#L115

How about re-enable the dynamic scaling?

Nanoseb, 2018-06-06:
Just implemented a 1st version of the background map, it works quite well, but it needs an internet connection...
I am planning to do a separate script that will, given a usual waypoint parameter file, download the .png of the map, and the live 2d plot will just display the png if available.

Nanoseb, 2018-06-07:
this is done with  3bddc80 , I just have to write some doc about it.