Maritime-Robotics-Student-Society/sailing-robot #252: Exclude manual control and land preparation from MarkSuccess()

tsaoyu, 2018-06-08:
In our first (17:58:40 07June 2018) and second run (18:34:40 07June 2018) the boat keep tacking because most attempts on tack failed when it was not in autonomous model. I assume jibe is always pick due to all tack fails when we either in manual control or doing preparation on the land. 

The issue ca be part addressed by  relaunch in the middle of ocean. 

It would be nice to exclude those attempts by reading the state of manual/autonomous.

Nanoseb, 2018-06-08:
Oh yes, I didn't think about that, that's indeed needed

Nanoseb, 2018-06-08:
Do you think resetting the history when ever we switch from remote control to autonomous is a good idea?

tsaoyu, 2018-06-08:
I would suggest down/up/down<sup>1</sup> combination in 1 second  as a signal to reset the history. The advantage is that short take over time wouldn't completely wipe out the history.  

<sup>1</sup> `up`:  AUTO or 0  `down`: MANUAL or 1