Maritime-Robotics-Student-Society/sailing-robot #253: Ubuntu version

sjdsm, 2018-07-15:
When we installed the needed dependencies for the sailing robot
 ~/sailing-robot/utilities/setup_scripts/Setup_ALL.sh workstation
It said that because our version of python is too low (2.7.6), we failed to install tornado and some other things related to it. Updating python costed a lot of time and still didn't work.

Solution:
The Ubuntu version we used was 14.04 which offically uses python 2.7.6. In deed,  now the package is based on Ubuntu 16.04 and ROS kinetics (with python 2.7.12). Changing our system successfully solved all those problems.

Nanoseb, 2018-07-15:
Great, thank you for your input, and welcome in the sailing robotic world (if you are new ;) ). 
If you are following our wiki tell us if you find other errors or unclear sections, we will be happy to help and improve it.
We are trying to update it, you can follow the progress here: https://github.com/Maritime-Robotics-Student-Society/sailing-robot/issues/236

tsaoyu, 2018-07-16:
@sjdsm Thanks for keeping promise and sharing the output in our discussion here!