Maritime-Robotics-Student-Society/sailing-robot #261: Xsense position investigating

CamilV, 2018-10-19:
The Xsense is the sensor that reads the boat's heading. Currently is placed on top of the Pi, below the battery, where the electromagnetic interference is quite bad (recalibration of the Xsense is required before every launch). It might be possible to move in a position in which electromagnetic interference is not as bad so that calibration would not be required so often

livid-fpv, 2018-10-20:
Conformal coat it and put it higher up on top of the mast?

CamilV, 2018-10-20:
On the top of the mast we already have the GPS and the wind sensor, there's quite a lot of magnetic interference up there. Conformal coat might be something worth considering

ReFil, 2018-10-20:
put it at the rear of the boat on a little post maybe?