Maritime-Robotics-Student-Society/sailing-robot #263: 1 battery to power the entire boat

CamilV, 2018-10-19:
Currently, the Pi is powered with a USB Powerbank and the servos are powered from 4 AA batteries. I'm thinking, powering the whole boat from a single 2S LiPo or something close to that, might be possible and it would facilitate test preparation and simplify the wiring

CamilV, 2018-10-20:
Why would you recommend a 4S? The Pi requires a power supply of 5V and the servos work well with voltage levels as low as 4.5V. My idea was to step down the voltage with a buck converter to power the Pi, ensuring a constant 5V supply and have another buck converter to power the servos.  The maximum current we saw being drawn by the servos was about 800mA, which would result in almost no voltage drop across the battery.

ReFil, 2018-10-20:
If something goes wrong and you need to keep power up you have more margin
to discharge a 4s below its rated voltage maintaining 5v for the supply. A
2s would work, but for stuff I do, I always have a nice big margin. also if
you use an 18650 pack for better energy density you'd need at least a 3s as
18650 cut off voltage is around 2.5v per cell depending on the cells you
choose