Maritime-Robotics-Student-Society/sailing-robot #265: Fish finder reverse engineering

CamilV, 2018-10-19:
Not required for the competition, but would be nice to reverse engineer a fish finder to send information such as water temperature, pH, depth, salinity and anything else that might turn our boat into a research vessel.

Something like this:
https://www.beschoi.com/05a00042r-smart-portable-depth-fish-finder-with-100-m-wireless-sonar-sensor-echo-sounder-fishfinder-for-lake-sea-fishing.html?currency=GBP&gclid=CjwKCAjwgabeBRBuEiwACD4R5tuUjSt1yE4aN6T0N_eR--W2IGAVlr6ZCgdqljVMylQ6iPVd9XjQwRoCCikQAvD_BwE

livid-fpv, 2018-10-20:
Most fishfinders do not have pH or salinity probes, they typically have only water temperature and depth, but there are many that speak publicly known protocols, usually NMEA 0182 or NMEA 2000, so getting the data that they do provide isn't impossible

smaria, 2018-11-09:
interesting that this fish finder would come up here :) I don't think it records anything but water temperature. Recordings were made with an SDR and this fish finder. However, in the swimming pool the recordings were made in, it appears the surrounding noise was louder than the ping of the fish finder itself. Ping me if anyone ends up looking into this (or wired fish finders).