Maritime-Robotics-Student-Society/sailing-robot #271: No Map received rviz

brunopinto900, 2019-03-30:
Hello,

I launched Rviz and open config.rviz with rosrun tf static_transform_publisher 0.0 0.0 0.0 0.0 0.0 0.0 map my_frame 100.

Rviz opens, but says "no map received"

[image]

tsaoyu, 2019-04-03:
Does the command `rosrun tf static_transform_publisher 0.0 0.0 0.0 0.0 0.0 0.0 map my_frame 100` came from the `grid_map` tutorial? I have searched our Wiki but none of them mentioned about this command.

brunopinto900, 2019-04-04:
Yes.
Here is the link.

https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/2D-plot-of-the-boat

tsaoyu, 2019-04-04:
We recommend the use of Matplotlib based 2D for the simulator.

Nanoseb, 2019-04-04:
Sorry I forgot to respond here. Indeed as tsaoyu said, the matplotlib plot is the one to prefer.

Concerning rviz, as specified in the wiki you need to install and setup rviz_satellite for it to display a map. But you can actually see the boat and waypoints etc. without the map, everything else should work fine.

brunopinto900, 2019-04-04:
I ran matplot 2D, everything is just fine.
I tried to work with rviz and failed, so i won´t try anymore.
How can i see the waypoints? I only see 2 points when running matplot 2D, is that it?

Thanks for your support.