Maritime-Robotics-Student-Society/sailing-robot #275: Archive of old sailing logs

takluyver, 2019-11-15:
If anyone's still working on the sailing robot project and could use it: I have an archive of old data recorded from sailing the Black Python in 2016, 2017 and 2018, including a mixture of test runs and competition runs at those years' WRSCs. I'm happy to hand this on.

The data collected includes:

- ROSbags: a log of all ROS messages during the run, can be replayed in ROS tools
- GPS traces: CSV files with GPS coordinates (there are scripts/notebooks in the repo which can plot these on a map)
- Parameter dumps: JSON files recoding all the [parameters](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/tree/master/src/sailing_robot/launch/parameters) in use each time we launched ROS.

This totals about 1.7 GB zipped, and 8 GB decompressed.

takluyver, 2019-12-20:
I discussed with @smaria the possibility of publishing this data on [Zenodo](https://zenodo.org/) or some similar archive. I've prepared a README for it, which I'll paste in below.

Would anyone rather I don't do this? Do you think another data repository would be better than Zenodo? Do you want to suggest changes to the README? If there are no objections, I'll publish this some time after Christmas.

-----
Readme text:

# Sailing Robot Logs

These files contain recorded data from the [Southampton Sailing Robot Team](https://www.sotonsailrobot.org/)'s boat, the *Black Python*. The logs cover various tests, and three [WRSC](https://www.roboticsailing.org/) competitions: 2016 (`viana`), 2017 (`horten`) and 2018 (`calshot`).

Each test run generated several files:

- `.bag` and `.bag.metadata`: [Rosbag](https://wiki.ros.org/Bags) files, recordings of messages sent among parts of the control system, which was built in the ROS 1.x framework. ROS includes tools to inspect and replay these files.
- `params-dump-*.json`: JSON dumps of the parameters used to launch that run, including waypoint GPS coordinates and sensor calibration parameters, among other values.
- `gps-trace*.csv`: GPS coordinates of the boat during the run, in the format required by WRSC. The three columns are timestamp (`hhmmssdd`), latitude and longitude both converted to integers by multiplying decimal degrees by 1e7, i.e. shifting the decimal point 7 places to the right. All GPS coordinates use the WGS-84 reference.
  - `gps-raw-nmea*`: Raw NMEA 0183 protocol data sent by the GPS receiver. This should be redundant with the CSV data so long as it was parsed correctly.
  - `map-gps-trace*.html`: GPS traces plotted in interactive maps using [Folium](https://python-visualization.github.io/folium/), overlaid on OpenStreetMap tiles.

`index.html` contains a generated overview of the test runs performed, and the files associated with each.

The source code used to create these files, including both the control system for the boat and scripts for processing the data is available in a git repository at:
https://github.com/Maritime-Robotics-Student-Society/sailing-robot

Exactly what data is in the different runs varies as the hardware and software were evolving at the same time. We did not record which commit generated each file.

takluyver, 2020-01-03:
I've started preparing an upload on Zenodo. Can everyone who was involved with the team from 2016-2018 let me know your preferred

- Name
- Affiliation
- ORCID (if you have one)

I never know whether affiliation should be now or from when the work occurred.

Alternatively, if you'd prefer not to be associated with the dataset at all, let me know that too. You can comment here or email me.

takluyver, 2020-01-14:
@tsaoyu @Nanoseb @NanaOKA @danhaus @CamilV @alvaropp - please let me know by the end of January if you're happy to be listed as an author on the dataset. If so, let me know your preferred form of your name, affiliation and ORCID, if you have one.

I'm happy to include anyone who was involved in the team 2016-2018 in the author list, even if you feel your contribution was minor.

I'm probably forgetting some people, so if you're in contact with anyone else from the team, please pass the message on to them. Sophia already offered to contact Pier & Elisavet.

danhaus, 2020-01-14:
Hi Thomas, I'm happy to be listed as an author (Daniel Hausner). Please let me know if you need any help with this. Many thanks

takluyver, 2020-01-15:
Thanks Daniel. Do you want me to put anything for your affiliation?

Nanoseb, 2020-01-15:
So sorry for the (way too) late reply, that's indeed a quite good idea, I didn't know these kind of things existed. For me:

- Sébastien Lemaire
- University of Southampton
- [0000-0002-9959-2100](https://orcid.org/0000-0002-9959-2100)

If you have any other question or need anything else, feel free to ping me, I will be more responsive.

CamilV, 2020-01-15:
Hi Thomas,

I'm also happy to be listed as author,
For me:
Camil Vasilovici
University of Southampton

Many thanks,
Camil

takluyver, 2020-01-15:
Thanks, I've ensured you're both on there.

danhaus, 2020-01-16:
> Thanks Daniel. Do you want me to put anything for your affiliation?

I guess you can put University of Southampton, since I studied there while working on the project. Thank you

tsaoyu, 2020-01-18:
My current information

- Yu Cao
- University of Edinburgh
- 0000-0001-5767-7029

Thanks

takluyver, 2020-01-19:
Thanks @tsaoyu! The form asks for names as "Family name, given names". Would that be "Cao, Yu"?

Sorry if this is a daft question. I know that Chinese names put the family name first, but I've also seen that some people reverse their name to use in European languages, so I don't know which version you use.

tsaoyu, 2020-01-19:
@takluyver 
Yes, Cao Yu is correct.

NanaOKA, 2020-01-22:
Hi Thomas,

Apologies for the late reply and this is a great idea. Please find my details below

Nana Abankwa
0000-0002-0090-3694

Thanks

takluyver, 2020-01-22:
Thanks Nana, that's great. Do you want any affiliation recorded? It's not required if you don't.

pierbiag, 2020-01-31:
Hi Thomas, so sorry to be the last ones!

- Pier Maria Biagiolini
- University of Southampton
- 0000-0001-6435-1094

- Elisavet Hadjiodysseos
- University of Southampton
- 0000-0003-2419-5335

Are you guys all ok?

takluyver, 2020-01-31:
Thanks both! I've added you on Zenodo. There are still a few more files I need to upload, then I think it's ready.

I'm very well, thanks. Well, today, I'm a bit depressed about brexit, of course, but good other than that.

takluyver, 2020-02-02:
Thanks everyone for sharing your details. The dataset is now published at: https://zenodo.org/record/3597449 . I can still update the metadata if there's some mistake.

I've uploaded each file separately - that's over 1200 files. I don't think zenodo has a 'download all' button, but you can automate downloads using the API, e.g. with [this zenodo_get tool](https://gitlab.com/dvolgyes/zenodo_get).

If anyone wants a full copy, I'm still happy to send a zip/tar file directly, but now that it's online, I'll delete the files at some point.

Nanoseb, 2020-02-02:
Awesome, thanks a lot Thomas for handling this. I have updated this repo readme to add a link to the zenodo page.

takluyver, 2020-02-13:
I found a few more files I'd overlooked before (from the Calshot obstacle avoidance), so there's a new version in Zenodo to include these. I've updated the README link.

https://zenodo.org/record/3666184

closed 2020-02-13 by takluyver